openhwfoundation/core-v-mcu #331: Build error when running `make sim`

LiamSkirrow, 2025-02-14:
Hi there,

I've been making some progress with getting the build/simulation environment up and running and I'm currently trying to run a simulation to check that everything works. However, I'm running into an error that I can't seem to figure out, any pointers would be helpful!

The error message is included below and it seems similar to this issue (#182)  , but I'm not sure if it ever got resolved. As mentioned in my comment on that issue, my Ubuntu version is `Ubuntu 24.04.2 LTS`.

My command history is very straightforward
```
cd core-v-mcu
make clean
make sw
make buildsim
make sim          # I get the error here
```
**_Error message:_**
It almost seems like a Verilog syntax issue, but I'm not sure if the error message is misleading me. It indicates that there is an issue with line 1 of the file `axi_id_prepend.sv`

```
> ~/Documents/code/core-v-mcu-socs(master)$ cat /home/liam-skirrow/Documents/code/core-v-mcu-socs/build/openhwgroup.org_systems_core-v-mcu_0/sim-modelsim/transcript 
# Reading pref.tcl
# //  Questa Sim-64
# //  Version 2021.2_1 linux_x86_64 May 15 2021
# //
# //  Copyright 1991-2021 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  QuestaSim and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
# vsim -gui -sv_lib ../../../tb/uartdpi/uartdpi -voptargs="-O1 +acc" core_v_mcu_tb 
# Start time: 21:26:44 on Feb 14,2025
# ** Note: (vsim-3812) Design is being optimized...
# ** Warning: (vopt-14005) Vopt is compiled with optimization level -O0 or -O1. The simulation might run slowly.
# vopt_stacktrace.vstf written
# ** Fatal: Unexpected signal: 11.
# ../../../rtl/vendor/pulp_platform_axi/src/axi_id_prepend.sv(1): Vopt Compiler exiting
# ** Note: (vsim-12126) Error and warning message counts have been restored: Errors=1, Warnings=1.
# Error loading design
# End time: 21:26:45 on Feb 14,2025, Elapsed time: 0:00:01
# Errors: 1, Warnings: 1

```

Thank you!

MikeOpenHWGroup, 2025-02-15:
Hi @LiamSkirrow, I see that you are using Questa on Ubuntu 22.04.  I was unaware that Questa supported Ubuntu - we access Questa from a CentOS machine.  So that _may_ be your Issue.  I'll poke around.

MikeOpenHWGroup, 2025-02-15:
Hi @gmartin102, if memory serves (and it never does), you are using VCS, is that correct?   Have you ever seen an Issue like this?  Do you have access to Questa?

LiamSkirrow, 2025-02-15:
Thanks for looking into this Mike, it's much appreciated. That's a good point regarding the potential compatibility issue with Ubuntu <-> Questasim, and it could indeed be the underlying issue. 

It appears that some other people have experimented with running qsim on Ubuntu before. I'll also have a poke around and see if I can make some progress and will update here with my findings 👍

gmartin102, 2025-02-16:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Mike,
I have run Questa-sim on ubuntu in the past, so I don’t think that’s the issue,  but currently we are using redhat and it runs fine for me, I did notice that  I suppress vlog-2583 in my scripts to run, but that should be the same as the make buildsim; make sim from the repo

Here is a snippet from core-v-mcu.core  --- the Makfile also does the run-gui invocation from the build sim directory so I get the waveform view up.

Greg

sim:
    <<: *default_target
    default_tool: modelsim
    filesets_append:
    - not_emulation-rtl
    - rtl-simulation
    - mem_init_modelsim
    - rtl-behavioral
    - tb-rtl
    toplevel:
    - core_v_mcu_tb
    parameters:
    - USE_BW=true
    tools:
      modelsim:
        vlog_options:
        - -override_timescale 1ns/1ps
        - -suppress vlog-2583
        vsim_options:
        - -sv_lib ../../../tb/uartdpi/uartdpi -voptargs="-O1 +acc"
      xcelium:
        xrun_options:
        - -64bit
        - -timescale 1ns/1ps
        - -disable_sem2009
        xmsim_options:
        - -sv_lib ../../../tb/uartdpi/uartdpi.so

dave-roo, 2025-02-20:
I have tested this on centos 9 and it did not work. The questasim version is 2021.1. 

@gmartin102 which version are you using?

Same error as @LiamSkirrow.

I'm unable to gather any additional info from questa regarding this.

Perhaps a tb of just the pulp platform axi module.

gmartin102, 2025-02-20:
I am running questa 2021.4
Also verified that build-xcelium runs properly xcelium version 20.09-s018
Greg

dave-roo, 2025-02-24:
I went into core-v-mcu/rtl/vendor/pulp_platform_axi/scripts/

and ran the vsim_run.tcl to test the axi module and did find an error:

Reading pref.tcl

```
# 2021.2_1

# vsim -sv_seed 0 tb_axi_dw_downsizer -t 1ps -gTbAxiSlvPortDataWidth=16 -gTbAxiMstPortDataWidth=8 
# Start time: 11:45:49 on Feb 24,2025
# ** Note: (vsim-3812) Design is being optimized...
# vopt_stacktrace.vstf written
# ** Fatal: Unexpected signal: 11.
# /home/vboxuser/Documents/core-v-mcu/rtl/vendor/pulp_platform_axi/src/axi_demux.sv(1): Vopt Compiler exiting
# ** Note: (vsim-12126) Error and warning message counts have been restored: Errors=1, Warnings=0.
# Error loading design
Error loading design
# End time: 11:45:50 on Feb 24,2025, Elapsed time: 0:00:01
# Errors: 1, Warnings: 0
```

MikeOpenHWGroup, 2025-02-24:
It is difficult to help with this one given that the tool is emitting a `** Fatal: Unexpected signal: 11.` error.  That sounds like a tool and/or platform problem.  It could be due to a problem with the MCU RTL or scriptware, but Questa should not fatal-out like that.  Can you reach out to your vendor for support?

dave-roo, 2025-03-06:
Hi, 

We have installed a newer version of questasim 10.7c and it has resolved the issue. We were able to successfully run "make sim", thanks for your support.

MikeOpenHWGroup, 2025-03-06:
Excellent!  I'll close this issue.