openhwfoundation/core-v-mcu #332: [Help] Compiler choice. Default gives error.

dave-roo, 2025-03-12:
### Is there an existing core-v-mcu bug for this?

- [x] I have searched the existing bug issues

### Bug Description

Which compiler is validated to work? 

We are trying to run the testbench and everything is working except the questasim default compiler gcc-5.3.0 gives a 
"Bad value, collect2".

Thanks

[image]

MikeOpenHWGroup, 2025-03-12:
Any reason you are using gcc 5?   That is pretty ancient.

If you could attached the entire logfile and provide the command lines you invoked (starting from `git clone...`) that would be helpful.

dave-roo, 2025-03-13:
All good, we moved to the later version and it worked.

closed 2025-03-13 by dave-roo